B-AROL-O/FREISA #19: [FEAT]: Print 3D part(s) to connect Mini Pupper 2 with LEGOs

gmacario, 2024-03-09:
### Description

From Afreez message on Discord group "general" of Pupper@MangDang, 2024-02-19:

[image]

gmacario, 2024-03-09:
After a quick message exchange over discord I agreed with Afreez the following process

[image]

In summary:

* The STL have already been published by @MangDangroboticsclub on Google Drive: https://drive.google.com/drive/folders/1TTmcGOG-JoAWNTGy1r7hbb7txhZIF4_p
* If someone needs to modify them, the updates will eventually be shared as PR to https://github.com/B-AROL-O/FREISA

gmacario, 2024-03-13:
And here is a photo of the first attempt made by @OrsoEric to 3D-print those parts using his [Creality K1 3D Printer](https://www.creality.com/products/creality-k1-3d-printer)

[image]

gmacario, 2024-03-16:
Here is a second 3D-print attempt, this time with a [Yellow PLA](https://www.amazon.it/gp/product/B07QQBY6PL)

[image]

gmacario, 2024-03-16:
However the best results have been achieved with this Red PLA

[image]

gmacario, 2024-03-16:
Here is our FREISA doggo after attaching the LEGO parts to it

[image]

closed 2024-03-16 by gmacario

mangdangroboticsclub, 2024-03-16:
Cool！

gmacario, 2024-03-16:
> Cool！

Indeed!
You know, @OrsoEric is **THE** "robot-guy" 😄