B-AROL-O/FREISA #47: [BUG]: ci: duplicate runs of the same jobs

davmacario, 2024-10-07:
### Description of the bug

Right now, the CI pipeline is triggered, either:

- By a commit on non-main branches
- By a PR

This results in duplicate jobs execution whenever pushing to a branch for which a PR is open.

My suggestion would be to deactivate the pipeline execution on push, or to create some logic to prevent duplicate jobs for when a commit is pushed on a branch that has a PR open.
I think the first solution makes more sense, as the current job only includes super linter.

If it will be required to add more jobs, then we can review the triggering rules.

### Steps To Reproduce

1. Create a new branch
2. Push some changes on the new branch
3. Open a PR from the new branch to main
4. Do an additional commit on the same branch

This^ will result in a total of 4 jobs (1 on point 2, 1 on point 3, and 2 on point 4), while the expected number is 3 (point 4 should only trigger 1).

gmacario, 2024-10-07:
> @[davmacario](https://github.com/davmacario) davmacario assigned [gmacario](https://github.com/gmacario) [1 hour ago](https://github.com/B-AROL-O/FREISA/issues/47#event-14535444809)

Thanks for the assignment @davmacario, however I am OK if anyone else (you?) can propose a fix.

davmacario, 2024-10-07:
Yep, my mistake, I wanted to assign this to myself

PR #48 (2024-10-14): [draft] ci: fix duplicate jobs

Closes #47  

## 📑 Description

- Prevent duplicate jobs - now Super Linter is only ran on pull requests
- Create new job to build and push Depthai Docker image to the GitHub Container Registry

## ✅ Checks

- [x] My pull request adheres to the code style of this project
- [x] My code requires changes to the documentation
- [ ] I have updated the documentation as required
- [ ] All the tests have passed

## ℹ Additional Information

The PR is still in draft, as I am trying to find a way to test the new job (I am accepting suggestions).

I also would like to trig [... cut]
closed 2024-10-14 by gmacario