B-AROL-O/FREISA #50: [FEAT]: Document how to use the Docker image published on GHCR

gmacario, 2024-10-14:
### Description

Now that [`github/workflows/publish_image.yaml`](https://github.com/B-AROL-O/FREISA/blob/main/.github/workflows/publish_image.yaml) has been pushed to the main branch, Docker images are published on https://github.com/orgs/B-AROL-O/packages?repo_name=FREISA

[image]

I would suggest to create a `docker-compose.yml` or at least a HOWTO document which explains how to use those images

gmacario, 2024-10-14:
@davmacario would you like to take ownership of this issue?

davmacario, 2024-10-14:
that's okay, I'll do it in the next days when I have the time