B-AROL-O/FREISA #51: [BUG]: Automatically built container images are not publicly available

gmacario, 2024-10-14:
### Description of the bug

FYI @davmacario I noticed that the Docker images published under https://github.com/orgs/B-AROL-O/packages?repo_name=FREISA are not available users not logged into GitHub:

https://github.com/orgs/B-AROL-O/packages?repo_name=FREISA

### Steps To Reproduce

1. Go to https://github.com/orgs/B-AROL-O/packages?repo_name=FREISA as logged in user "@gmacario"
2. Verify that the container images are available
3. Access the same site in incognito mode and check agaoin

### Additional Information

I am not sure whether this is intentional or due to wrong configuration of the repository, though

davmacario, 2024-10-14:
I think the answer lays in [here](https://docs.github.com/en/packages/learn-github-packages/configuring-a-packages-access-control-and-visibility) somewhere

gmacario, 2024-10-15:
> I think the answer lays in [here](https://docs.github.com/en/packages/learn-github-packages/configuring-a-packages-access-control-and-visibility) somewhere

Indeed. More specifically, the details are listed at https://docs.github.com/en/packages/learn-github-packages/configuring-a-packages-access-control-and-visibility#configuring-visibility-of-packages-for-an-organization:

> **[Configuring visibility of packages for an organization](https://docs.github.com/en/packages/learn-github-packages/configuring-a-packages-access-control-and-visibility#configuring-visibility-of-packages-for-an-organization)**
>
> When you first publish a package, the default visibility is private and only you can see the package.<br>
> You can grant users or teams different access roles for your package through the access settings.<br>
> Once you make your package public, you cannot make your package private again.
>
> 1. On GitHub, navigate to the main page of your organization.
> 2. Under your organization name, click the **Packages** tab.
> 3. Search for and then click the name of the package that you want to manage.
> 4. On your package's landing page, on the right-hand side, click **Package settings**.
> 5. At the bottom of the page, under "Danger Zone", click **Change visibility** and choose a visibility setting:
>    * To make the package visible to anyone, click **Public**.<br>
> Warning: Once you make a package public, you cannot make it private again.
>
>    * To make the package visible to a custom selection of people in your organization, click **Private**.

gmacario, 2024-10-15:
> Indeed. More specifically, the details are listed at [...]

Since this operation cannot be reverted, before making the image public for everybody I would first suggest to implement a fix to #50 then have some "friend customer" (i.e. @pitdagosti, or @alv67) independently test it.

WDYT @davmacario?

gmacario, 2025-09-18:
The issue is still present in v3.1.1:

```bash
gmacario@delta:~$ docker pull ghcr.io/b-arol-o/freisa:v3.1.1
Error response from daemon: Head "https://ghcr.io/v2/b-arol-o/freisa/manifests/v3.1.1": unauthorized
 1 gmacario@delta:~$
```

alv67, 2025-10-03:
> [@alv67](https://github.com/alv67) would you be able to help us test the (still private) Docker image https://github.com/B-AROL-O/FREISA/pkgs/container/freisa before we make it publicly available?

Pull is working, but I tryed by using a Access Token with all authorization enabled.

alv67, 2025-10-03:
> > [@alv67](https://github.com/alv67) would you be able to help us test the (still private) Docker image https://github.com/B-AROL-O/FREISA/pkgs/container/freisa before we make it publicly available?
> 
> Pull is working, but I tried by using a Access Token with all authorization enabled.

Works with Access Token authorized for repository only

[image]